florianfesti/boxes #291: Pattern generator

matkoniecz, 2021-03-02:
Is it reasonable/feasible/in scope to add also applying patterns to boxes?

Either something to make them prettier or for providing tactile feedback for blind people (or in dark / out of eyesight for some other reason).

I am asking as I am starting to make project to generate tactile patterns and I am unsure is there some reasonable way to integrate it with this project.

matkoniecz, 2021-03-02:
Or is it something out of scope and if I want something like that I should implement it separately?

Maybe there should be a way to pass pattern specification into boxes.py with pattern specified outside this project? (maybe there is already way to do this and I missed it)

usecase: I am making a gift for my sister and I am sure she will prefer box with pattern over plain one.

florianfesti, 2021-03-02:
I'd say pattern are in scope and could at some point get their own group. For now there is no good way to have them added directly onto the boxes. But there might be in the future. I have some ideas for bundling all callbacks of a box into one object that could then be applied onto a box. Such pattern might make a nice use case.
For now I'd suggest to just have a generator per pattern with x, y parameters. If you want to make them more box friendly you could add an z parameter that makes the generator render all six sides if a non zero value is given.

florianfesti, 2021-03-02:
Oh, be aware that boxes.py might not be a great basis to implement intricate patterns as the graphics capabilities are very limited and barely enough for rectangles

matkoniecz, 2021-03-02:
In that case I will try to implement it as a separate project for now and try to keep it reasonably configurable.

I will post something here in case of producing something useful (if you know about such existing project - feel free to mail me about it to email visible in my GitHub account)

closed 2021-03-02 by matkoniecz