florianfesti/boxes #294: Use with CNC Mill -> inhibit corner loops

jenshackel, 2021-03-03:
BOXES is generating small loops at 90 degree corners (maybe to ensure better parts removal?). This is perfect for laser cutting where the "tool" laser is extremly thin. I'd like to use boxes with my cnc mill where the tool is some mm thick. The loops are causing unwanted holes and give overall bad milling results. Is it possible to disable / inhibit the "corner loops" feature?

florianfesti, 2021-03-03:
Sure, just set the `burn` value to zero. You will need to do the kerf correction on your own though. Your CAM software should be able to handle that.
There is no way to get kerf correction without those loop in Boxes.py right now. With the new graphics back end one probably could something about that now but there are a few issues like not all inner corners actually being replaced with the loop as shown above but still being the original reversed arcs (see slightly outdated https://florianfesti.github.io/boxes/html/api_burn.html )

florianfesti, 2021-03-03:
Oh, the reason for these loops is that you can't  cut sharp corners with a round tool. They make sure that there is enough room for the part filling the corner.This is actually not really needed for soft material like wood when cutting with a tool as small as a laser. Depending on the diameter of your milling bit you may need those loops much more than a laser cutter. Look up "dog bone cnc"

jenshackel, 2021-03-03:
Thanks for your ultrafast reply! It's working now as wanted ;-)
Sure, the extended corners are really important - especially for CNC - but when cutting some mm small joints with a 3mm diameter tool is causing ugly holes. But my main issue is that I am using "dxf2gcode" as converter and the loops can't be processed (somtimes the converter even crashes).
Thanks for your support and - a really phantastic tool for creating boxes !

closed 2021-03-03 by jenshackel