florianfesti/boxes #295: Pictures help, but how about examples to help filling out fields?

vinito, 2021-03-13:
ie.: Traylayout... 
I was trying to figure out how to format the field (default shown below for 3x3):
_________________________
 ,> 0.0mm
 | ,> 0.0mm
 | | ,> 0.0mm
+-+-+-+
| | | | 0.0mm
+-+-+-+
| | | | 0.0mm
+-+-+-+
| | | | 0.0mm
+-+-+-+
_________________________
The photo for the generator shows a (1+2) x 2 which is similar what I want to make, but try as I might, I cannot figure out how to format this field to generate anything other than full evenly spaced dividers. Everything I try returns "An error occurred!"
It would be handy to be able to reference an example format for tricky options like what I'm looking for. Since there's a photo already showing a divider that not only has one full-width row with one divided row, PLUS a bottom panel, I for one think it would be really helpful just to look at what formatting generated the divider in the photo. The picture is nice and all, but the example would be magnitudes more helpful to me. Just my $.02

florianfesti, 2021-03-13:
Yes, the TrayLayout urgently needs some description text. Here's an example layout for a 2+1 box:
```
 ,> 40.0mm
 | ,> 40.0mm
+-+-+
| | | 70.0mm
+-+-+
|   | 60.0mm
+-+-+
```
Note that the bottom middle wall has been replace by a space character.

florianfesti, 2021-03-15:
Added a description text to both steps of the generator.

commit 7397e49b (2021-03-15): TrayLayout: Add description text

Resolves: #295
closed 2021-03-15 by florianfesti