florianfesti/boxes #297: Default silverware generates self-intersections

matkoniecz, 2021-03-16:
https://festi.info/boxes.py/Silverware?language=en press generate 

Note 
[image]

florianfesti, 2021-03-17:
> There are actually quite a few generators that create self-intersections and other quirks on default settings. And it is rather inconvenient when you want to get a general idea of the design without fiddling with the settings.

Please open tickets!

florianfesti, 2021-03-17:
Yeah, the Silverware generator needs some love. That's not the only issue. It probably needs a top part similar to the RoundedBox. It is one of the oldest generators but hasn't made it out of "Unstable" yet.

florianfesti, 2021-09-21:
OK, at least the most urgent fixes...

commit f6fa4967 (2021-09-21): surroundingWall: Fix size calculation

Resolves: #297
closed 2021-09-21 by florianfesti