florianfesti/boxes #300: Rounded box vertical part has parts that are too tightly fitting

matkoniecz, 2021-03-17:
Rounded box vertical part has parts that are too tightly fitting. For reference, I used https://festi.info/boxes.py/RoundedBox?FingerJoint_angle=90.0&FingerJoint_style=springs&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=3&FingerJoint_play=0.0&FingerJoint_space=3&FingerJoint_width=1.0&Flex_stretch=1.05&Flex_connection=1.0&Flex_distance=0.5&Flex_width=5.0&x=120.0&y=120.0&h=100.0&outside=0&outside=1&radius=20&wallpieces=1&edge_style=F&top=lid&thickness=3.65&format=svg&tabs=0.0&debug=0&reference=0&burn=0.13&render=0

I consulted with someone who made more than one successful box and their advice was too modify generated output manually - as it generates too tight fit by default.

Probably it would be better to provide some space there automatically.

My burn parameter worked nicely in burn test and other part of the box, but I managed to destroy that part - and still failed to push pieces together. I tried pushing it with fingers, rubber hammer (mallet?) and clamp that managed to leave visible shape on wood and failed to push pieces together.

It applies to left and right part of

[image]

[image]

florianfesti, 2021-03-18:
There is only one value for kerf correction that is used everywhere (`burn`). If those dove tails don't fit you need to use a lower value. If you feel the finger joints are too soft try using the "rectangular" variant which is stiffer.

matkoniecz, 2021-03-18:
I just hoped/expected that one burn parameter will fit all designs. Different box worked perfectly with springs fitting equally tightly.

florianfesti, 2021-03-18:
There really is not much more a box generator can do. Finding the right burn value requires experimentation and experience. Every laser cutter and material is different. And even the same material requires different settings depending on thickness.

matkoniecz, 2021-03-18:
In this case problem is that burn is well fitting springs/fingers but not fitting kerf. And according to what other person using laser told me - it is normal that kerf correction is too tight.

bwente, 2021-03-19:
It also looks like you are working against the grain. You might have better luck with MDF or acrylic. Or try steaming the wood.

florianfesti, 2021-06-03:
Basically testing the spring type finger joints won't transfer to other joint types. There's no way to fix that. May be adding a parameter to the spring type finger joints to adjust their fit might improve the situation. But I feel this is kinda out of scope here. Let's just close this for now.

closed 2021-06-03 by florianfesti

matkoniecz, 2021-06-03:
Is there a way to have two different joint types, requiring different burn parameters, in one design?

Without applying manual fixes to generated design?

florianfesti, 2021-06-03:
No, the burn correction is part of the lowest level drawing primitives (`Boxes.corner()`). See https://florianfesti.github.io/boxes/html/api_burn.html if you are really interested in the details. Where a different fit is needed the features need a setting on their own. The spring variant of finger joints not having one is the real issue here. Places that need looser fit typically have a `play` parameter or just offer more space in the geometry.

matkoniecz, 2021-06-03:
So for [my original design](https://festi.info/boxes.py/RoundedBox?FingerJoint_angle=90.0&FingerJoint_style=springs&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=3&FingerJoint_play=0.0&FingerJoint_space=3&FingerJoint_width=1.0&Flex_stretch=1.05&Flex_connection=1.0&Flex_distance=0.5&Flex_width=5.0&x=120.0&y=120.0&h=100.0&outside=0&outside=1&radius=20&wallpieces=1&edge_style=F&top=lid&thickness=3.65&format=svg&tabs=0.0&debug=0&reference=0&burn=0.13&render=0) I should use higher burn value and negative play for finger joints, right?