florianfesti/boxes #312: SlatWall - 100mm versus 3” distance between slats

ChrisLaserman, 2021-04-16:
Love the slatwall boxes, but a setting to adjust the distance between top hook and bottom hook would be great. It appears to be hard-coded for standard European 100mm slatwall.  And there are no US sources for 100mm, just 3”.  Can you add an adjustable setting for slat distance?  Or hard coded 76.2mm?  I have no python skills.

Or is there a configuration setting that I am missing?
Thanks!

florianfesti, 2021-04-17:
Yes, these are still a bit rough. Many generators have been only ever used once. It's also possible that the hooks are a bit too tight for some walls. I will need to have a look at them soon anyway as we got a new make of slatwalls in our workshop and I need to redo some of our tool holders.
So try out some smaller pieces before pouring too much laser time and material in your builds.

closed 2021-04-17 by florianfesti

florianfesti, 2021-04-18:
Added `hook_depth` in the slat wall settings to make the hooks fit into shallower slits.

commit 29183ed7 (2021-04-17): SlatWalls: Make pitch of the wall adjustable

Resolves: #312