florianfesti/boxes #313: Burn Value on boxes.py for universal box not working

tperrycb, 2021-04-28:
This just started in the last couple months since I cut some boxes, can't nail the time exactly, but I thought I had made a mistake after I cut some boxes it was very tight.  So I recut the test pattern from .05 to .12   .05 is very sloppy and .12 is a snug fit.   However I set the universal box to .1 and .11 in two different tests and it was extremely tight, to the point that you almost can't get it together.   Then I cut a box at .05 and it had the exact same fit of extremely tight to almost not usable.   I made no changes in the cutting software ( using lightburn ) between cuts. 

The test pattern did seem to work though with different burn settings.

florianfesti, 2021-04-28:
Mistakes can happen. But this seems very unlikely to be a bug as the burn value is very fundamental and part of the basic drawing primitives.

Can you please attach the rendered file or just check yourself. The fingers should be two times `burn` wider than their nominal width while the spaces in between should be two times the `burn` narrower. Note that this is middle to middle of the lines. The lines should be the width of two time `burn`. So the fingers and the spaces in between should be just their nominal size between the lines.

If in doubt I often open the file in Inkscape and create a rectangle with the nominal size of a finger that is filled but without a border to avoid any issues from line width. This can then be dragged around the drawing to check if everything fits.

Be aware that the burn setting depends on many things. Different material, thickness, laser cutters, etc need different burn values. Wrong focus height, poor beam alignment and differences in air assist can all influence the kerf of the machine.

But I have to admit that a tight fit with 0.05 requires a pretty good machine. So may be there is an software issue after all...

tperrycb, 2021-04-28:
Ya.  I will give it a shot.  What I was focusing on was the test pattern worked fine, that was generated with the 2 squares, .05 is very sloppy, and .12 is snug but good.   So I then used the same settings in on the universal box and it didn't work and the fit was exactly the same regardless of what I put in the burn field.     I went through recent changes and didn't see anything that stood out were it seems like that would have got broken, but haven't had time to study the code in depth.

ffesti, 2021-05-04:
Yes, please. Please provide all other information you have - like whether you downloaded the dxf directly from Boxes.py or converted with an external tool, and the link you used to generate the file (if you still have it). The more I know the easier it is to find the cause.

Thanks!

florianfesti, 2021-05-22:
I have not found anything. If you still have a file showing the breakage please re-open and attach the file. Thanks!

closed 2021-05-22 by florianfesti