florianfesti/boxes #317: Thank you!

Feredyr, 2021-05-06:
Hello,
I wanted to share with you what I build thank to your work.
Here Is my first insert for the boardgame "It's a wonderful world"
[image]
the dividerTray to store all cards
[image]
And the small closed boxes for tokens.
[image]
Those ones are  3 different closed boxes melt in Inkscape to get the lid  with large fingers and cut edges of the sides.

again Thank you :)

Bye

florianfesti, 2021-05-16:
Very nice work!

As soon as I get #234 done I need to bug you for the parameters to merge the three boxes/trays into one generator.

florianfesti, 2021-06-03:
You are welcome! Seeing Boxes.py put to good use like this is what keeps me going.

Still closing this here as I try to keep the ticket list short. You are invited to post this project in a comment on the Hackaday.io project page (https://hackaday.io/project/10649-boxespy) where it is probably much more visible.

closed 2021-06-03 by florianfesti