florianfesti/boxes #318: Errors when building Inkscape Extensions under Windows 10

alastaira, 2021-05-08:
Hi, using latest code from respository, and Python 3.9.5, I am able to run the local boxes.py webserver, but having problems compiling the extensions for Inkscape following the guide at https://florianfesti.github.io/boxes/html/install.html#inkscape

The first issue seems to relate to the way that the setup script attempts to concatenate variables and handling of spaces. By default Windows installs Python in the "C:\Program Files\Python" directory, but calling "python setup.py build" results in the error "'C:\Program' is not recognized as an internal or external command, operable program or batch file." because it has a space in the directory name.
Changing line 20 of the setup script to include single quotes around the system executable fixes this, i.e. os.system("'%s' %s %s" % 

However, the next problem is that Line 25 attemps to call "xgettext", which is not a windows command. I'm not sure what it's purpose is, or what a suitable replacement is, but is there any other way to build the Inkscape extensions under Windows? Thanks!

alastaira, 2021-05-08:
Ok, so I found that even though setup.py won't execute, I could still run scripts/boxes2inkscape to generate the .inx files and manually copy them to the extensions directory. Now, this doesn't add the necessary directories to the PATH though, so I've added in boxes, boxes/boxes, and boxes/scripts, and now the extensions menu appears in Inkscape with no errors, although none of the generators actually create anything in the Inkscape document either....

alastaira, 2021-05-09:
Making some progress... it seems the .inx files are hardcoded to look for a dependency on the "boxes" executable placed directly within the inkscape extensions folder which is also where they need to reside. 
For those people who have lots of extensions, that's a bit unwieldy - it would be helpful to contain all the boxes scripts within their own subfolder of the extensions directory (which is a feature newer versions of Inkscape have supported for some time).

I think the next problem relates to Windows permissions not allowing the "boxes" main script to execute from the extensions folder though. I can only get it to work from the command line when placed in, say, c:\boxes (but when I do that, the inkscape extensions don't find it...).

If there is a setup known to work with Inkscape under Windows10, it would be great if you could share the expected final location of all the necessary files, and the PATHs thatnewd to be defined. In the meantime, I'll keep tweaking!

florianfesti, 2021-05-17:
You may want to have a look at #314 and the linked instructions at https://stadtfabrikanten.org/display/IFM/Boxes.py

florianfesti, 2021-05-22:
Closing this here as #314 looks more likely to fix this issue and spreading this topic on multiple places does not really help

closed 2021-05-22 by florianfesti