florianfesti/boxes #327: Erreur Export .DXF

jubonhomme, 2021-06-25:
Exemple sur cette boite depuis la version en ligne ( https://www.festi.info/boxes.py) 
[https://www.festi.info/boxes.py/TrayLayout2?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&h=50.0&hi=0.0&outside=0&outside=1&layout=+%2C%3E+50.0mm%0D%0A+%7C+%2C%3E+30.0mm%0D%0A+%7C+%7C+%2C%3E+30.0mm%0D%0A+%7C+%7C+%7C+%2C%3E+30.0mm%0D%0A+%7C+%7C+%7C+%7C+%2C%3E+30.0mm%0D%0A%2B-%2B-%2B-%2B-%2B-%2B%0D%0A%7C+%7C+%7C+%7C+%7C+%7C+50.0mm%0D%0A%2B-%2B-%2B-%2B-%2B-%2B%0D%0A%7C+%7C+%7C+%7C+%7C+%7C+50.0mm%0D%0A%2B-%2B-%2B-%2B-%2B-%2B%0D%0A&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&burn=0.08&render=0](https://www.festi.info/boxes.py/TrayLayout2?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&h=50.0&hi=0.0&outside=0&outside=1&layout=+%2C%3E+50.0mm%0D%0A+%7C+%2C%3E+30.0mm%0D%0A+%7C+%7C+%2C%3E+30.0mm%0D%0A+%7C+%7C+%7C+%2C%3E+30.0mm%0D%0A+%7C+%7C+%7C+%7C+%2C%3E+30.0mm%0D%0A%2B-%2B-%2B-%2B-%2B-%2B%0D%0A%7C+%7C+%7C+%7C+%7C+%7C+50.0mm%0D%0A%2B-%2B-%2B-%2B-%2B-%2B%0D%0A%7C+%7C+%7C+%7C+%7C+%7C+50.0mm%0D%0A%2B-%2B-%2B-%2B-%2B-%2B%0D%0A&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&burn=0.08&render=0)

L'export en .SVG fonctionne bien mais pas en .DXF
Voici le message d'erreur (sur Firefox ou Chrome depuis Ubuntu 20.04)
Une erreur s'est produite !
Conversion failed. pstoedit returned 1

florianfesti, 2021-06-25:
Yes, easily reproduced. Looks like the meta data in the Postscript output is not properly quoted. Should be easy to fix. In the mean time please use the SVG output and convert it to dxf  with Inkscape or any other tool.

closed 2021-06-25 by florianfesti

jubonhomme, 2021-06-25:
Yes I still use Inkscape to convert SVG to DXF.
But I have a problem with the DXF transformed by Inkscape in the laser cutter software.  (RDworks)

florianfesti, 2021-06-25:
Should work now.

jubonhomme, 2021-06-25:
Yes it works perfectly

commit 9cb038ef (2021-06-25): Quote cli in meta data to not contain line breaks

Resolves: #327