florianfesti/boxes #332: Universal Box Output Not Consistent

ralxnder, 2021-09-30:
I create a universal box with no lid. I changed the material thickness to match my material and changed my burn rate to match what I've used before (determined through the burn test). I also changed the bottom edge to use the Finger Joint. I then save the output as a dxf, pdf and svg. The dxf and pdf finger widths are the same but the svg finger widths are different. Although I believe the SVG is probably more correct.
[universalBox.zip](https://github.com/florianfesti/boxes/files/7261218/universalBox.zip)

florianfesti, 2021-10-09:
That's kinda weird. I opened both the SVG and the PDF in Inkscape and the fingers are the same in both. Looks like the PS output (which is converted to PDF and DXF by `pstoedit`) doesn't have the optimization for inner corners. But the finger sizes look right to me. May be the software you are using are doing weird things on import. Or I am missing the spot where things are wrong.

Can you set `reference` and check if it has the right size. It does not have any burn correction so setting it to 100 should result in a rectangle of exactly 100mm length.

For all formats the assumption is that the laser cutter cuts "on the line" - so in the middle of the width. And the width of the line is what is supposed to get burned away. So it's width is twice the `burn` value.

florianfesti, 2021-10-09:
OK, after reading your bug report again you may just be talking about the inner corners being different. This is indeed an issue that needs fixing. Looks like the line width (that is used whether the arc in the corner is small enough) is not properly set.

closed 2021-10-09 by florianfesti

ralxnder, 2021-10-09:
Understood. I double checked and it seems to be working fine now. Thanks.

commit fce7e08a (2021-10-09): Adjust line width in transform()

Pass scale param down the transform call tree.
This fixes the inner corners for the PS output which were seen as "too
big" when comparing with the wrong line width.

Resolves: #332