florianfesti/boxes #337: Question: interested in code formatting?

javl, 2021-11-05:
Before I start making any changes I was wondering if you'd be interested in a reformatted version of boxes.py, using something like the pycodestyle linter. No change in functionality, just an update to use f-strings and some more consistent formatting.
Also, a bunch of files currently don't have a file extension (especially in the scripts folder). While this is valid on Linux, it is kind of odd and it would be nice to add them.

florianfesti, 2021-11-08:
Although the project is Python 3 nowadays most of the code base very much is still Python 2. Also a lot of pieces were done rather quickly. So reformatting and converting to a more modern style is welcome. This should obviously be done in separate commits that don't do functional changes.
File extensions are a bit of a different topic. They are frowned upon for executables on Unix systems. So I'like to keep the executable names there. But I am open to install them with extension on Windows or may be even rename them but add symlinks on Unix.

javl, 2021-11-08:
Do you have any references for this last comment, on file extensions being frowned upon? I've never seen this in any other project (been running Linux for years).

florianfesti, 2021-11-08:
OK; may be frowned upon is a bit too harsh. A look into /usr/bin shows there actually are a few executables that use file extensions. But most don't - even when only looking at scripts. Guess this has become more acceptable in recent years -  with Python being the worst offender. I generally think a file extension leaks an implementation detail into the (CLI) interface. Which is something to be avoided.

Otoh as the name of the project is `Boxes.py` I should probably reconsider my POV...

florianfesti, 2023-01-05:
Guess the answer is "Not as much as I first thought"

closed 2023-01-05 by florianfesti