florianfesti/boxes #344: tote box with centre handle

Azzman77, 2021-11-23:
Forgive me if I am missing it or if its already been raised, but is there a generator for a multi compartment open top box with the ability to add a centre partition the extends up with a handle? I know this option is available[image] to do[image] this on the back of the box but not the middle. cheers
[image]

florianfesti, 2021-11-24:
No, there currently is no generator to do this. After thinking a bit about it I don't think I will add one. There are just too many possible variants to get this nicely into the UI. Just use TypeTray or TrayLayout and modify the piece(s) you want handles on with Inkscape or any other vector graphics program.

closed 2021-11-24 by florianfesti

Azzman77, 2021-11-24:
I would think it's an option that would be more popular than having this
exact option for the back. And simpler because the sides wouldn't change at
all.