florianfesti/boxes #346: Keypad parts overlapping

gcollic, 2021-11-27:
Hello 👋

I noticed that the keypad triangle parts are overlapping (with default settings).

[image]

The code generating them is:

``` python
        self.rectangularTriangle(
            tr, tr, "ffe", num=4,
            callback=[None, lambda: self.hole(trh, trh, d=d1)]
        )
```

https://github.com/florianfesti/boxes/blob/master/boxes/generators/keypad.py#L92-L95

Jikstra, 2021-11-29:
Same for the electronicsbox, seems that the rectangularTriangle method with the edges argument "ffe" is causing the problem.

florianfesti, 2021-12-01:
Yeah, no idea how I messed that up.

commit ad1be4c3 (2021-12-01): Fix placement of rectangularTriangle(s)

That was broken by fcefddca1e5a9b18c92991e7451ef5891f69109f. Turns out
the the handling of self.spacing (distance between parts was not handled
properly.

Resolves: #346

Thanks to Guillaume Collic for reporting!
closed 2021-12-01 by florianfesti