florianfesti/boxes #347: Wrong size with flangedWall

RaphaelDives, 2021-12-02:
Hello!
I very like this project and I hope I can give you back something in the future! :)

But... I'm trying to create a flangedWall and a normal rectangularWall like this:

`self.rectangularWall(150, 50, "ffEf", move="right",label="test1")`
`self.flangedWall(150, 50, "ffEf", flanges=[0,0,10,0], r=5, move="right",label="test2")`
`self.flangedWall(150-2*self.thickness, 50, "ffEf", flanges=[0,0,10,0], r=5, move="right",label="test3")`

So the first test is "correct" width, the second test is 2* thickness wider and when I correct this with a smaller width, the fingers are not the same anymore... please help!

Cheers
Raphael

florianfesti, 2021-12-03:
Yeah, the flangedWall was hacked together a bit quickly for being used in the ElectronicsBox. Basically it does not support the use of "f" edges. Switch them out for "F" and things should work out.
But yes, it should check for legal edge types or handle them more gracefully. I may have a look at that on the week end.

florianfesti, 2021-12-03:
To elaborate a bit further: The "F" edges ca be replaced by finger holes if there is a flange protruding out. There is currently no hole pattern for an "f" edge. So the flangedWall is meant as a base plate in which all other walls slot in with an "f" edge. Note that the other supported edge type is "E" and not "e" as it is also outset by one thickness. There should be support for "e" edges, too but currently isn't. (Actually "e" edges should work but will also be outset).

RaphaelDives, 2021-12-04:
Thank you very much!
If I COULD read I could have found out myself, since it's written in the source file...
I found a temporary workaraound using a compound edge. ;)
In the meantime I will try to understand how boxes are working, but although python is not new to me, understanding a complex project like this, takes it time...

florianfesti, 2021-12-04:
OK, I fixed the code to work with other edge types if there is no flange at that side. This may break your code but leads to more sane results.

commit ee2baab3 (2021-12-04): flangedWall: Allow other edge types

for sides without flange. Support "e" edges for sides with flanges
(along with "E" and "F")

Resolves: #347
closed 2021-12-04 by florianfesti