florianfesti/boxes #348: BinTray middle 'floor' is misplaced when sy has different widths

Al0rin, 2021-12-04:
[For example 80:20](https://festi.info/boxes.py/BinTray?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=0.5&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&sx=50&sy=80%3A20&h=50&outside=0&outside=1&front=0.4&thickness=1.5&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=0&burn=0.1&render=1):

I think the middle floor should rather be placed to the indicated (red) position:
[image]

Am I right? If not just close the ticket

Al0rin, 2021-12-04:
This is what the model looks like when modified: [image]

florianfesti, 2021-12-04:
Yes, this is obviously wrong. It is surprisingly difficult to come up with test cases that are not symmetrical in any direction...

commit b11f9b0f (2021-12-04): BinTray: Fix finger holes in side walls

Pattern was mirrored before.

Resolves: #348

Thanks to https://github.com/Al0rin for reporting!
closed 2021-12-04 by florianfesti