florianfesti/boxes #349: Dependencies on affine and Python 3.6

gadgetmind, 2021-12-06:
Hi, and thanks for this great application.

I had problems using it initially because -
1) The code uses fstrings, so needs at least Python 3.6. It just threw a syntax error for me, but I was stupid and initially focused on the ImportError that this caused. Maybe the code could check early on that the Python version is suitable? Other features could be used elsewhere and throw runtime errors.
2) I already had affine on my Ubuntu install (I guess installed with apt getting the python3-affine package) and this generated broken svgs with lines too long and odd squiggles. Removing this and using pip3 to install affine fixed it instantly, so I guess there is also a dependence on affine version.

florianfesti, 2021-12-12:
Yeah, the `setup.py` surely can be improved. Do you have a version number for the affine versions that do and do not work?

gadgetmind, 2021-12-12:
Doing an apt search suggests that if I reinstalled python3-affine on my Ubuntu Xenial box I'd get "python3-affine/xenial,xenial 1.2.0-1"

Looking at my site packages I see - python3.7/site-packages/affine-2.3.0

This is what it looked like with the "bad" one.

[image]

I hope this helps.

florianfesti, 2021-12-12:
From the change log of 2.0-beta:

- Breaking change: we have reversed our sense of rotation, a positive angle
  now rotates a point counter-clockwise about the pivot point (#25).
- Bug fix: a bug in matrix-vector multiplication had been reversing the
  direction of rotation and is now fixed (#25).

I guess requiring affine>2.0 is probably a good idea...

commit a593e8e8 (2021-12-12): setup.py: Require Python>=3.6 and affine>=2.0

As older versions won't actually work

Resolves: #349

Thanks to https://github.com/gadgetmind for reporting!
closed 2021-12-12 by florianfesti