florianfesti/boxes #351: A PaperBox that needs less paper

fuddl, 2021-12-17:
I tried to recreate a chocolate box with the dimensions 85×84×13 mm. Unfolded, this  box could be made out of a sheet of 133×199 mm.

[image]

If I generate a PaperBox [with the same dimensions](https://festi.info/boxes.py/PaperBox?x=84&y=13&h=85&lid_heigth=15&lid_radius=7&lid_sides=20&margin=0.5&mark_length=1.5&tab_angle_rad=0.07982998571223732&finger_hole_diameter=15&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&burn=0.1&render=1) it needs sheet of 213×294 mm with a lot of paper to throw away after. Also if scaled up, it exceeds the maximum paper size of my plotter. 😅

I propose a more space efficient PaperBox design based on my chocolate box.

Box                   | sheet size needed
---------------|---------------------
PaperBox         | 62 622mm²
[new design]  | 26 467mm²

The new design requires 42.26% less paper. Is it a good idea? Is that something worth looking into?

gcollic, 2021-12-17:
Hello !

I'm the original contributor of this paper box. Here is my opinion ;-)

Is the original paper box like you want if you exchange y and h ? https://festi.info/boxes.py/PaperBox?x=84&y=85&h=13&lid_heigth=15&lid_radius=7&lid_sides=20&margin=0.5&mark_length=1.5&tab_angle_rad=0.07982998571223732&finger_hole_diameter=15&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&burn=0.1&render=1

Your box is similar but doesn't look identical.The current box was created for advent calendar cases, it's not clear if I can use your design for this purpose without some tradeoff, and I wonder if your design is more efficient with any parameters combination (in paper size, but also folding simplicity, and opening ergonomy).

I think having both could maybe be interesting. I'm not sure if it would be better as a whole new box or as a simple "design type" parameter for the current box. In both cases, we could think collectively of how to name both so that the difference makes sense ;-)

gcollic, 2021-12-17:
On the sample I linked with y and h switched, I'm happy with the "tabs" which takes half of the length of the box, so that they are touching each others when folded, since it doubles the thickness of the paper when glued, but if this sample is similar to your target maybe the tab length is the missing parameter we could add to optimize in your case ?

fuddl, 2021-12-18:
hi, @gcollic thanks for your quick feedback.

> Is the original paper box like you want if you exchange y and h?

Not quite, the important feature of the chocolate box (which I forget to mention) is that it opens on the narrowest side, rather then the _top_ side. It is also designed to be unglued on the tabs since they are part of the opening mechanism:

[image]

I think  [on the image above](#issue-1083486287) it's clear where the glue used to be.

I propose to rename the parameters (or rather the labels of the input fields) to make more clear which side the lid is. Or provide some clues like a sketch:

[image]

> maybe the tab length is the missing parameter we could add to optimize in your case ?

Yes, that sounds great.

fuddl, 2021-12-18:
The other parameter seems to be: are the tabs supposed to be glued or are they part of the lid mechanism?

[image]

gcollic, 2021-12-18:
Thank you for all the explanations, I think I understand better.

Your box have 2 lids on the narrowest side without cutout for the finger (but I suppose it could), when mine has only one on the top side with a cutout for the finger.

Both have tabs to be glued : 4 glued against walls in my design like your right diagram, 2 glued to each others in yours (not visible in your diagrams).

Both have tabs that are part of the opening mechanism (2 for each lid), so both our box are like your left diagram.

I think it's a totally different design, but I'm still not sure if a second box or a "design type" option makes more sense.

If they are different boxes, maybe a "flat paper box" and a "regular paper box" ?

I'm wondering if @florianfesti has an opinion on the subject ^^

gcollic, 2021-12-18:
>provide some clues like a sketch

I love your sketches by the way! Very clear!

florianfesti, 2021-12-18:
I don't have a strong opinion on the subject. My initial instinct was that having variant in the same generator is probably better UI wise. But that might just be me being bad with generator names and not being too much interested in paper boxes in the first place. So I am fine with both approaches. Having separate generator may make it easier to start an Paper group if more paper based generators emerge.

gcollic, 2021-12-19:
Thank you @florianfesti :-)

@fuddl are the following differences right and comprehensive? (black = original only, red = new only, green = both)
[image]

There is a lot of differences, but parameters could all be the same (dimensions, optional round cut for thumbs, tabs length, etc.).
I think I would keep a single box with a parameter for the kind of box (top box and tuck box ?). And in the `render` method I would directly switch between two render sub-methods dedicated to each kind, reusing the same parameters and helper methods (finger_hole, etc.).

What do you think ?

PS: I think I will use your design for other purposes also, such as game cards tuck box ^^

fuddl, 2021-12-19:
This looks great and comprehensive. I would personally not need a second lid → only one of the openings needs a thumb cut.

I like the single parameter approach but it should be made clear what the parameter does. I'd be happy to provide explanatory sketches if you want.

Also I'm  curious to see, the diff in the code, if you decide to implement it.

florianfesti, 2022-01-03:
Guess this is done! Should have added a "Resolved" line to the commit message.

closed 2022-01-03 by florianfesti