florianfesti/boxes #353: PaperBox with folding lines

fuddl, 2021-12-28:
My plotter supports debossing which can be used for drawing folding lines rather than outside folding marks. I'd like to draw them in a different color or on a separate layer or both:

[image]

the problem is, I don't know how to draw different colored lines and I don't know how to move to a different position without drawing a line (like the [MoveTo SVG command](https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/d#moveto_path_commands)).

florianfesti, 2021-12-29:
There's [Boxes.moveTo()](https://github.com/florianfesti/boxes/blob/master/boxes/__init__.py#L1057).

You can use `self.set_source_color(Color.ETCHING)` (or `self.set_source_color(Color.GREEN)`) to set the color for folding lines. Color handling is still not in its final form but this is what we got so far and it should work for you.

fuddl, 2022-01-09:
@florianfesti @gcollic could you please show me how you would integrate a second path in `paperbox.py`? I would supply the exact geometry then.

florianfesti, 2022-01-09:
The way the PaperBox generator is currently implemented makes it very hard to add such lines. What you need to do is stop at the right place and insert the drawing code for the crease line there and then continue with the outline.
Assembling a long list  to be passed to `polyline` make this pretty annoying. So the first step would be to refactor the code to a procedural style where methods don't return a list but draw a piece of the box themselves. After that one can easily add the crease lines to the parts that need them.

After having a quick first try I would rather not do the refactoring myself but I can provide the code that needs to be added to draw the crease lines. It should be something along the line of (untested)

> 
    self.polyline(...) # line up to the place where the crease line needs to go
    with self.saved_context():
        self.set_source_color(Color.ETCHING)
        self.moveTo(0, 0, 90)
        self.edge(tab_length)
    self.polyline(...) # contine outline

fuddl, 2022-01-12:
I tried this. The line is drawn but it is part of the same path and wont have a different colour 🤷

florianfesti, 2022-01-12:
Ahhh, there is `self.ctx.stroke()` missing to make the crease line a separate path. I guess adding it at the end of the indented block should do the trick. If I only could remember why I think that...

fuddl, 2022-01-12:
@florianfesti okay, that works. but the black path will be split into segments each time, I draw a green line. I think I'm fine with this, but does it cause problems on your side?

florianfesti, 2022-01-12:
Quite possible that the (fairly new) back end needs a bit more love. Just ignore those issues for now. If I care enough I'll fix that myself later on.

florianfesti, 2022-07-01:
This commit should solve the issues with lines not being the right color if they are the continuation of the outline

florianfesti, 2023-01-11:
No activity for a year. I am closing this now. Feel free to re-open or post a PR if you continue on this project.

closed 2023-01-11 by florianfesti