florianfesti/boxes #354: Eurorack

KittyDreadful, 2021-12-29:
It seems like there should be at least three variable for this box - width, depth and height.  There are only two: _inner height in mm_ and _Width of the case in HP_.  

When I created a file using inner height = 400 and Width of the case in HP = 10 (this is just to experiment) - the box was actually  134.7mm x 403.2 x 57mm.  The 403.2 value makes sense, but where did 134.7 and 57 come from?  

Also, the picture provided shows vents.  This box has no vents.

florianfesti, 2021-12-29:
The distance between the rails is fixed to the Eurorack standard (see https://en.wikipedia.org/wiki/Eurorack). So there is no third dimension needed. The length of the rails is measured in HP (0.2inches or 5.08mm). So 10HP are 50.8mm inner width resulting in a 56.8mm box when including 3mm walls (57mm when including two time the 0.1mm burn correction aka kerf).

The description of the box obviously could be better as it assumes people are familiar with the Euorack format.

The ventilation holes don't come with the generator but have to be added manually (as in all other enclosures). Guess I should finally add a few more to the `Hole Patterns` group...

closed 2021-12-29 by florianfesti

KittyDreadful, 2022-01-01:
Thanks for the explanation!