florianfesti/boxes #355: Feature Request

KittyDreadful, 2022-01-01:
Hi,
I've reviewed the existing settings in UniversalBox for the slide-on lid box. 

I'd like to suggest another design type for this type of box.

Here's an example:
https://3axis.co/download/91jdwd41/

It's a regular box with additional attributes:
1. insets that make the railing (back, left and right)
2. shortens the front panel to accommodate the slider lid (which makes the slot) for the lid
3. the slider lid 

Please let me know if this is feasible.  Thanks.
kitty

florianfesti, 2022-01-02:
Have a look at the CardBox which is very similar: https://festi.info/boxes.py/CardBox?language=en

KittyDreadful, 2022-01-02:
I looked at the card box, but it looked the top was hinged.  When I generated it, the lid has finger  joints.   I don't understand why the lid would have joints.

florianfesti, 2022-01-02:
Hmm, looks like people have issues with this box - even after adding part labels, a picture of the details of the slot and lid and assembling instructions.
There is a slim piece that gets added to the lid to form a grip and to complete the top ridge around the lid.
I fixed the description in  fdc86189003945f2f559b87484a828316dd149ab which will go online soon. Any suggestions on how to explain this more clearly?

florianfesti, 2022-01-05:
Improved the description a bit more in 0b3a0506d7d8e4469571ebab0208914d5d9fb377 (not yet online).
I am closing this ticket as I don't see anything left to do. Feel free to re-open if there are still some features missing.

May be you can provide a better replacement picture for the whole box if you cut one yourself.

closed 2022-01-05 by florianfesti

KittyDreadful, 2022-01-06:
> Improved the description a bit more in 0b3a050 (not yet online).

As soon as it's not below freezing here, I can cut and photograph a new picture.   Please let me know if I can help with the description as well.