florianfesti/boxes #356: Feature Request: Open Topped Box

bbum, 2022-01-05:
There is "OpenBox", but that eliminates two sides.  And "ClosedBox" which has all sides.

I want "OpenBox" but with just the "top" missing.

And I'd like a "make lid" option that creates a lid whose ID is the same as the OD of the open box plus, say, 0.2mm or so.   And the lid's height should be adjustable so it can be just a lid and not a sleeve.

I.e. easily be able to generate one of these with or without the lid.

[image]
[image]
.

florianfesti, 2022-01-05:
There is the UniversalBox which comes in many forms. One of them is a simple box that is open at the top. It currently does not support this kind of lid out of the box (no pun intended) but you can create a second one manually. It already supports a few other lids so may be this one can added at some point - probably after implementing generator being usable within generators.

bbum, 2022-01-05:
Oh! Thanks!

The Universal Box is really hard to understand.   Too many knobs and it isn't obvious what they do.  Without examples of each, it is hard to figure out.   Having helped a bunch of folks in various laser cutter communities, UniversalBox has been the source of confusion.

I would suggest that a adding "open top" as an option to the ClosedBox option would be far more approachable.

The lid is easy enough to make once one figures out how to make an open box so it can definitely wait until the generator w/generators is working!

KittyDreadful, 2022-01-06:
The two piece box is very close, but not quite there.

florianfesti, 2022-01-06:
Well, if you can fly a 747 you should be quite familiar with the controls of the UniversalBox...

May be I really should add a SimpleBox generator - and add some more documentation to the UniversalBox.

florianfesti, 2022-01-07:
Here you go: https://festi.info/boxes.py/ABox?language=en

Be aware that requesting a new box generator comes with the duty to provide a good example picture.

There is already some machinery to select a lid that is used my the UniversaBox. But I am not that happy with the current sate of affairs. I may put this in an proper Settings group and then add a a lid as described above.

commit 70162bdf (2023-04-06): Move Lid details and creation into own LidSettings class

Add new ontop and overthetop lid styles

Replace _ChestLid mixin class. Currently supported for UBox, UniversalBox
and ABox

Resolves: #356
closed 2023-04-06 by florianfesti