florianfesti/boxes #358: Two possible features: lasercut piano hinges, and integral box latches

jarkman, 2022-01-09:
I've just made this box, starting with a boxes.py fingerjoint box and adding some features which worked well. I thought they might be interesting as possible enhancements. I am unlikely to find time to write the code myself.

Photos and a DXF are here:
[https://www.jarkman.co.uk/catalog/cnc/hotwirebox.htm](url)

The piano hinge is an idea I got from Martin Raynsford, just an etched groove in a pair of layers that hold a wire. Makes a surprisingly effective hinge. I think it would also work well with printer filament instead of wire.

The latch is a structure I designed, which makes an effective latch built into the existing faces of the box. It uses a long torsion beam to make a sprung latch. If you chamfer the edges of the latch part, it snaps closed in a nice way, and it opens when you press down on the latch.

My latch is symmetrical, with a spring on each half of the box, but it would also be possible to make it with a spring in only one side, which would reduce the minimum required box height.

florianfesti, 2022-01-10:
These are both really nice. I wanted to play with laser cut compliant mechanisms for quite a while. This latch is a really great example.

jarkman, 2022-01-10:
Thanks! I am also a fan of compliant lasercut things, it's nice to find an application where they work so tidily.

florianfesti, 2022-01-10:
In case you have not seen it I make use of laser cut springs at a few places. The most prominent place is at the Console2 - as described at the bottom half of https://hackaday.io/project/10649-boxespy/log/177879-new-boxes-for-your-electronics-projects

jarkman, 2022-01-10:
Nice! I love the long-beam latches.