florianfesti/boxes #359: RegularBox not working for n=7

ecraven, 2022-01-10:
I've tried creating a regular box with 7 sides (for the seven weekdays), but this gives me no tabs on the top/bottom heptagon. n=6 and below works fine.
Any ideas why this is?

```./boxes RegularBox --n 7 --h 100 --radius 25 --top hole --output /tmp/foo.svg --format svg --thickness 4 --outside=true```

closed 2022-01-10 by ecraven

florianfesti, 2022-01-11:
May be this generator should have better defaults for the Finger Joint Settings. You can reduce `surroundingspaces` in the Finger Joint Settings and - if this is not sufficient - the `finger` size. This wait you can squeeze at least one small finger even on short edges.

florianfesti, 2022-01-21:
9f4657cc64b9218938de71e6073f10c1db51577f sets a better default.