florianfesti/boxes #360: Different sizes within TrayLayout

Destroher, 2022-01-19:
Hi, first and foremost: my apologies if I do something wrong or double, it is my first time on Github. I'm not a coder by any means.

I was wondering if it is possible to adjust each "box" in TrayLayout separately, so it could like the tray on the left side in this picture: [www.readytoplaystorage.com/pics/201806251405482.jpg](photo)

Could be that I need to add additional steps and formatting in the layout form?

If it's not possible: not a problem!  I will figure something out ;-)

Thank you for this great, great tool. I'm not there yet to lasercut something but this tool has made it an absolute breeze designing stuff!

(How do I add the Label "Enhancement"?)

florianfesti, 2022-01-19:
This is possible but may be not in the way you expect. While the TrayLayout is always based on a full grid single walls can be deleted by replacing them by a space in the layout. To get a tray similar to the picture you'd use something like this:

 https://festi.info/boxes.py/TrayLayout2?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&h=60&hi=0.0&outside=0&outside=1&layout=+%2C%3E+60.0mm%0D%0A+%7C+%2C%3E+60.0mm%0D%0A+%7C+%7C+%2C%3E+30.0mm%0D%0A+%7C+%7C+%7C+%2C%3E+20.0mm%0D%0A+%7C+%7C+%7C+%7C+%2C%3E+50.0mm%0D%0A%2B-%2B-%2B-%2B-%2B-%2B%0D%0A%7C+%7C+%7C+++++%7C+50.0mm%0D%0A%2B-%2B-%2B+%2B+%2B+%2B%0D%0A%7C+%7C+%7C+++++%7C+50.0mm%0D%0A%2B-%2B-%2B+%2B+%2B+%2B%0D%0A%7C+%7C+%7C+++++%7C+50.0mm%0D%0A%2B-%2B-%2B+%2B+%2B+%2B%0D%0A%7C+%7C+%7C+++++%7C+20.0mm%0D%0A%2B+%2B+%2B-%2B-%2B-%2B%0D%0A%7C+%7C+%7C+++%7C+%7C+30.0mm%0D%0A%2B-%2B-%2B+%2B+%2B+%2B%0D%0A%7C+%7C+%7C+++%7C+%7C+20.0mm%0D%0A%2B+%2B+%2B-%2B-%2B-%2B%0D%0A%7C+%7C+%7C+%7C+++%7C+30.0mm%0D%0A%2B-%2B-%2B-%2B-%2B-%2B%0D%0A&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&burn=0.1&render=0

Don't worry about the complicated URL. This is just a x=5 y=7 layout with the details filled in.

closed 2022-01-19 by florianfesti

Destroher, 2022-01-19:
Thank you very much, that is exactly what I was looking for! Thanks for the insight, you saved me hours worth of troubles :-P

I don't know if I'm overlooking it, but you should add some kind of PayPal button to your project :-)

Again, thanks!
Kind regards
Anthony

florianfesti, 2022-01-19:
Yeah, it's on my todo list. You can use the one here in the mean time https://owba.de/unterstuetzen/spende/ Sorry, page is German only.