florianfesti/boxes #362: dxf?

josemariasilva, 2022-01-28:
how i can add more formats?

florianfesti, 2022-01-28:
For a local instance install pstoedit to get the same formats as the website.
The back end currently can create SVG and Postscript natively and convert to other format from that. If you can suggest new converters adding supported formats should be easy.
Adding support for generating new formats is a bit more work but can be done by adding another Surface class to https://github.com/florianfesti/boxes/blob/master/boxes/drawing.py

florianfesti, 2022-01-29:
To help here I need a bit more details on your setup. What OS? How and where was Boxes.py and pstoedit installed?

josemariasilva, 2022-01-31:
my operating system is windows 10, I installed pstoedit from http://www.calvina.de/pstoedit/ and put it in the variables environment. I can use it normally from the terminal, but it doesn't convert it says an error when I run this command "pstoedit -f "dxf: -ctl -mm" A.svg B.dxf".

florianfesti, 2022-02-07:
OK, guess you are the first one trying to get pstoedit work on Windows. As I am on Linux only I have not really tested this. Please report back if it does not solve the issue.

commit b8c8a9ec (2022-02-07): Fix pstoedit for non Unix OSs

Resolves: #362
closed 2022-02-07 by florianfesti