florianfesti/boxes #365: Donation links?

BinaryConstruct, 2022-02-09:
Hi, this generator is amazing and I have used it numerous times over the years. I would like to contribute, but python and photography are not really my strong suit. Do you have any donation links, preferred charities or other ways I could contribute?

Blancanegra, 2022-02-09:
It's not the first time I've thought the same!.

florianfesti, 2022-10-07:
Well, I am putting this off over and over again as it requires getting the legal stuff ironed out. In the mean time there is a very tiny, German only page at my local hackerspace that does have a Paypal link. But I have not tried that in quite a while.

https://www.hackerspace-bamberg.de/Boxes.py

Blancanegra, 2022-10-08:
Thank you!
I just made a donation, so could you confirm if it works now or not!

florianfesti, 2022-10-08:
Yes, that worked. Thanks for the generous donation and the kind words!

commit 07f61ad8 (2025-01-01): How to give something back

Add page and link to Backspace donation page

Resolves: #365
closed 2025-01-01 by florianfesti