florianfesti/boxes #366: Extend Slatwall edges to also work with DIN Rails

florianfesti, 2022-02-09:
The Slatwall generators could easily modified to attach to other systems - like DIN rails or may be even slatwalls. May be exening the Slatwall edges is the easiest way to get this done as it won't require changing any of those generators (hopefully).

florianfesti, 2022-02-11:
If first thought coming up with some drop in replacement edge classes fro the Slatwall edges may be the way to go. But as they work with the hooks poking through the back wall we need most of this logic for the DIN rail edges, too - if we want to use them in the same generators.

So I have some first code that adds a `type` parameter to the SlatwallSettings that replaces the slatwall hooks by a Din rail one. There are still a lot of open questions though, like 

* Where to locate the bottom spacer
* What other settings are needed? Rail height?
* Do we need some locking mechanism and how to reach that. It probably needs to poke out at the top of the box.

dangerbock, 2022-02-11:
- Where to locate the bottom spacer
   Sorry didn't understant

- What other settings are needed? Rail height?
   maybe this ones:
   Rail Height
   Rail Depth

- Do we need some locking mechanism and how to reach that. It probably needs to poke out at the top of the box.
  I think something like the attach 
[image]

florianfesti, 2022-02-11:
> * Where to locate the bottom spacer
>   Sorry didn't understant
The Slatwall edges do have a bottom hook that is located on another slot - if this is possible. For the din rail we need a spacer as far down as possible. But it must not collide with another rail - assuming they are mounted in a regular pattern. Another way of handling this would be to put the spacer a t the very bottom and may be offer an offset to move it up a bit if necessary.

> * Do we need some locking mechanism and how to reach that. It probably needs to poke out at the top of the box.
>   I think something like the attach

I am not a fan of that. First it may be obscured by other parts of the box/tool holder/whatever. Second it relies on the thickness of the back wall for strength. For laser cut stuff thin walls are preferable. I was thinking about having the hook at the bottom and some kind of latch at the top that can be opened from the top. Or may be having a hook on top and a spring loaded catch on the bottom that allows lifting the box up against spring tension and then unhook it on top first and then on the bottom.

Note that the hooks here are constructed from side walls that are perpendicular to the rail and not the way as in the picture.

dangerbock, 2022-02-11:
> The Slatwall edges do have a bottom hook that is located on another slot - if this is possible. For the din rail we need a spacer as far down as possible. But it must not collide with another rail - [...]

In my opnion the simple way is the second one, to put some spacer with the Depth from the rail. The option to have an offset seens to be nice

> I am not a fan of that. First it may be obscured by other parts of the box/tool holder/whatever. Second it relies on the thickness of the back wall for strength. For laser cut stuff thin walls are [...]

The hook solution seen to be simple ... the solution with a spring that you already have in the current dinrail template is more complicated.

florianfesti, 2022-02-11:
See 7eca5a6c050fd55ad244cebc8f5060c9dca0d380 (not online yet).

dangerbock, 2022-02-15:
> See [7eca5a6](https://github.com/florianfesti/boxes/commit/7eca5a6c050fd55ad244cebc8f5060c9dca0d380) (not online yet).

already installed on my pc, i will make a test cut...but for what i see the "hook" type don't do nothing

dangerbock, 2022-02-15:
for what i see if i select stud it may work :)

florianfesti, 2022-02-16:
Yeah, this is still pretty alpha. All the settings need more explanation on what type of wall they work. Hooks may not ever be supported on DIN rails.
I also have a few ideas on more DIN rail connections. May be I'll add them in addition.

commit d1bb0985 (2022-07-30): WallEdges: French Cleats

Resolves: #366
closed 2022-07-30 by florianfesti