florianfesti/boxes #369: Build error on Linux

lucky62, 2022-02-13:
I am trying to install Boxes to the Inkscape on Linux Mint but I am getting the error:

```
# scripts/boxes2inkscape /usr/share/inkscape/extensions/
Traceback (most recent call last):
  File "scripts/boxes2inkscape", line 158, in <module>
    b.writeAllINX(sys.argv[1])
  File "scripts/boxes2inkscape", line 152, in writeAllINX
    self.writeINX(name, box, path)
  File "scripts/boxes2inkscape", line 145, in writeINX
    f.write(self.generator2inx(name, box))
  File "scripts/boxes2inkscape", line 112, in generator2inx
    if self.arg2inx(a, prefix) != "":
  File "scripts/boxes2inkscape", line 62, in arg2inx
    for e in a.choices:
TypeError: 'NoneType' object is not iterable
```

The similar error when trying to build manually:
```
# ./setup.py build
running build
running build_py
updatePOT()
po/boxes.py.pot:4940:10: invalid control sequence
xgettext: found 1 fatal error
generate_mo_files()
buildInkscapeExt()
Traceback (most recent call last):
  File "scripts/boxes2inkscape", line 158, in <module>
    b.writeAllINX(sys.argv[1])
  File "scripts/boxes2inkscape", line 152, in writeAllINX
    self.writeINX(name, box, path)
  File "scripts/boxes2inkscape", line 145, in writeINX
    f.write(self.generator2inx(name, box))
  File "scripts/boxes2inkscape", line 112, in generator2inx
    if self.arg2inx(a, prefix) != "":
  File "scripts/boxes2inkscape", line 62, in arg2inx
    for e in a.choices:
TypeError: 'NoneType' object is not iterable

** (inkscape:25864): WARNING **: 17:19:12.467: Invalid option --system-data-directory
running build_scripts
# 
```
Some files are generated and copied into Inkscape extensions dir but seems incomplete because all submenus in inkscape are empty:
[image]

lucky62, 2022-02-13:
additional info - file boxes.py.Dispenser.inx is empty:

```
# ls -l /usr/share/inkscape/extensions/boxes.py.*
-rw-r--r-- 1 root root 4923 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.ABox.inx
-rw-r--r-- 1 root root 3947 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.AgricolaInsert.inx
-rw-r--r-- 1 root root 9820 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.AllEdges.inx
-rw-r--r-- 1 root root 5437 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.AngledBox.inx
-rw-r--r-- 1 root root 4373 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.AngledCutJig.inx
-rw-r--r-- 1 root root 4423 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.Arcade.inx
-rw-r--r-- 1 root root 3516 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.Atreus21.inx
-rw-r--r-- 1 root root 4630 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.BasedBox.inx
-rw-r--r-- 1 root root 2836 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.BayonetBox.inx
-rw-r--r-- 1 root root 4874 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.BinTray.inx
-rw-r--r-- 1 root root 2812 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.BottleStack.inx
-rw-r--r-- 1 root root 4406 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.BurnTest.inx
-rw-r--r-- 1 root root 4486 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.CardBox.inx
-rw-r--r-- 1 root root 3911 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.Castle.inx
-rw-r--r-- 1 root root 4636 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.ClosedBox.inx
-rw-r--r-- 1 root root 2653 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.CoffeeCapsuleHolder.inx
-rw-r--r-- 1 root root 3267 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.ConcaveKnob.inx
-rw-r--r-- 1 root root 5674 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.Console.inx
-rw-r--r-- 1 root root 6442 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.Console2.inx
-rw-r--r-- 1 root root 4697 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.DinRailBox.inx
-rw-r--r-- 1 root root 5171 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.DiscRack.inx
-rw-r--r-- 1 root root    0 feb 13 17:18 /usr/share/inkscape/extensions/boxes.py.Dispenser.inx
```

closed 2022-02-16 by florianfesti

lucky62, 2022-02-16:
Thanks for the fixing.
I pulled the change but issue still exists:

```
# ./setup.py build
running build
running build_py
updatePOT()
po/boxes.py.pot:4940:10: invalid control sequence
xgettext: found 1 fatal error
generate_mo_files()
buildInkscapeExt()
Traceback (most recent call last):
  File "scripts/boxes2inkscape", line 158, in <module>
    b.writeAllINX(sys.argv[1])
  File "scripts/boxes2inkscape", line 152, in writeAllINX
    self.writeINX(name, box, path)
  File "scripts/boxes2inkscape", line 145, in writeINX
    f.write(self.generator2inx(name, box))
  File "scripts/boxes2inkscape", line 112, in generator2inx
    if self.arg2inx(a, prefix) != "":
  File "scripts/boxes2inkscape", line 62, in arg2inx
    for e in a.choices:
TypeError: 'NoneType' object is not iterable

** (inkscape:31038): WARNING **: 22:40:18.454: Invalid option --system-data-directory
running build_scripts
# 
```

```
# python3 scripts/boxes2inkscape /usr/share/inkscape/extensions
Traceback (most recent call last):
  File "scripts/boxes2inkscape", line 158, in <module>
    b.writeAllINX(sys.argv[1])
  File "scripts/boxes2inkscape", line 152, in writeAllINX
    self.writeINX(name, box, path)
  File "scripts/boxes2inkscape", line 145, in writeINX
    f.write(self.generator2inx(name, box))
  File "scripts/boxes2inkscape", line 112, in generator2inx
    if self.arg2inx(a, prefix) != "":
  File "scripts/boxes2inkscape", line 62, in arg2inx
    for e in a.choices:
TypeError: 'NoneType' object is not iterable
#
```

Dispenser still not generated.

florianfesti, 2022-02-16:
Weird. Works for me. This looks like you have not checked the changes out. Can you check your boxes/generators/dispenser.py whether the patch is applied?

lucky62, 2022-02-16:
yes, patch is applied:
[image]

lucky62, 2022-02-17:
I did small workaround. The problem is that ELIF condition has two parts joined with OR.
If the first part is true then second part can be false (a.choices can be null), but code is expecting that it is not null. So I add a condition before the FOR loop (where a.choices processed) and script is passing.
[image]

lucky62, 2022-02-17:
Menus are still empty. I will create a new issue because it is a different problem.

commit 529574fb (2022-02-16): Dispenser: Fix sideedges parameter

Looks like both the ArgparseEdgeType machinery and the inx conversion
should be more robust...

Thanks to https://github.com/lucky62 for reporting!

Resolves: #369