florianfesti/boxes #370: Empty submenus in Inkscape->Extensions->Boxes.py

lucky62, 2022-02-17:
all submenus under the Boxes.py extension menu are empty:
[image]

I observed that in the INX files the script is set as boxes_proxy.py:

```
<script>
    <command location="inx" interpreter="python3">boxes_proxy.py</command>
</script>
```

but this file not exists:

```
# ls -l
total 100
drwxrwxr-x 3 lucky62 lucky62  4096 feb 16 22:29 boxes
drwxr-xr-x 4 root    root     4096 feb 16 22:36 build
-rw-rw-r-- 1 lucky62 lucky62    39 feb 16 22:29 CODE_OF_CONDUCT.md
-rw-rw-r-- 1 lucky62 lucky62  6634 feb 16 22:29 CONTRIBUTING.rst
drwxrwxr-x 3 lucky62 lucky62  4096 feb 16 22:29 documentation
drwxrwxr-x 2 lucky62 lucky62  4096 feb 16 22:29 examples
drwxrwxr-x 2 lucky62 lucky62  4096 feb 17 08:38 inkex
-rw-rw-r-- 1 lucky62 lucky62 35147 feb 16 22:29 LICENSE.txt
drwxrwxr-x 5 lucky62 lucky62  4096 feb 16 22:29 locale
-rw-rw-r-- 1 lucky62 lucky62   174 feb 16 22:29 MANIFEST.in
drwxrwxr-x 2 lucky62 lucky62  4096 feb 16 22:29 po
-rw-rw-r-- 1 lucky62 lucky62  3646 feb 16 22:29 README.rst
-rw-rw-r-- 1 lucky62 lucky62    34 feb 16 22:29 requirements.txt
drwxrwxr-x 2 lucky62 lucky62  4096 feb 17 07:44 scripts
-rwxrwxr-x 1 lucky62 lucky62  3977 feb 16 22:29 setup.py
drwxrwxr-x 4 lucky62 lucky62  4096 feb 16 22:29 static
# 
# find . -name "*proxy*"
# 
# find . -name "*quire*"
./requirements.txt
# 
```
 It is missing in the sources? Or should be generated during the build?
Is this the cause of empty submenus?

florianfesti, 2022-02-18:
Yes, yes it is. Looks like I hit merge a bit too quickly on the Inkscape 1.0 changes...

florianfesti, 2022-02-18:
These changes have been made to make Boxes.py run on MightyScape but it looks like that broke the standalone version. See
https://gitea.fablabchemnitz.de/FabLab_Chemnitz/mightyscape-1.X/src/branch/master/extensions/fablabchemnitz/boxes.py
boxes_proxy.py can be found there. But I have not (yet) examined if this is really needed or if it is better to switch back to directly use the `boxes` executable which still have some special code for this purpose (which is probably outdated and not working with Inkscape 1.0)

bringert, 2022-02-27:
In Inkscape 1.1 on MacOS, this instead seems to show all the submenus greyed out.

I worked around this by creating a symlink called boxes_proxy.py in the Inkscape extensions directory:

$ cd ~/Library/Application Support/org.inkscape.Inkscape/config/inkscape/extensions
$ ln -s ~/src/boxes/scripts/boxes boxes_proxy.py

florianfesti, 2023-02-26:
Yeah, I added the boxes_proxy.py to the Inkscape directory with 122d365914d9d9b9ede634b7f1e11688c32bd353

Somehow missed to update this ticket. SHould work now.

closed 2023-02-26 by florianfesti