florianfesti/boxes #371: po/boxes.py.pot:4920:10: invalid control sequence

dherakles, 2022-02-24:
Hi, thanks for your tool boxes py, I love using it to replace plastic with wood.
My Platform WIN10/64, Python39, Inkscape 1.1.
Despite several attempts, I can't get any further with the precompilation:

„C:\Program Files\Inkscape\share\inkscape\extensions\boxes>python setup.py install
running install
running bdist_egg
running egg_info
writing boxes.egg-info\PKG-INFO
writing dependency_links to boxes.egg-info\dependency_links.txt
writing requirements to boxes.egg-info\requires.txt
writing top-level names to boxes.egg-info\top_level.txt
reading manifest file 'boxes.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'boxes.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
updatePOT()
po/boxes.py.pot:4920:10: invalid control sequence
xgettext: es ist 1 fatal error occurred
generate_mo_files()
buildInkscapeExt()
error: [WinError 2] The system can not find the stated file."

I would be very grateful for a help.
Best Regards
D Berneker

warcat-mg, 2022-03-12:
Hi - Have you solved it ? I am having the same problem.  And looking for clues

florianfesti, 2022-03-12:
Ok, this should fix it. Turns out different parts of the code have different ideas what characters are allowed and what they really mean.

commit b12fded5 (2022-03-12): Remove backslash from description

as it trips up gettext.

Also update language files.

Resolves: #371
closed 2022-03-12 by florianfesti