florianfesti/boxes #373: measurements include material thickness + xyz cordinates

tomse, 2022-03-20:
Thank you for an awesome product :-D 

Though I do have some trouble with some smaller cosmetics:

On a page like this:
https://www.festi.info/boxes.py/StorageShelf

* cordinate system xyz:
x, sy and sh seems somewhat confusing also in relation to the helptext.

x should be depth or z
sy should be sw or sx
and sh is fine or should be sy
(taking a regular xyz cordinate system into consideration)

* measurements include material thickness
Using outer measurements and 3mm thickness
in the example data, sy = 3*50 
the end result will be a box of 153. 
3 + 47 + 3 + 47 + 3 +47 + 3

I've had a few trial and errors with this.

is it possible to print out the dimensions for the inner measurements without material thickness?
I know it's simple math, but when you play with the parameters and check/uncheck outer measurements, one can easily
run into cutting the wrong sizes.

florianfesti, 2022-03-20:
Yeah, this is basically a TypeTray that has been adjusted by a contributor. Overall coordinates are not used very consistently in Boxes.py but this is one of the worse offenders.

Unfortunately I have failed to reproduce your issue with the size of the box being unexpected. Can you post a link with the parameter that show the issue. The default values result in a 150.4 x 150.4mm back wall, which seems fine to me for `3*50` in both directions and `outside` selected.

tomse, 2022-03-20:
Sorry, I wasn't clear on that.
It was more to have it "human readable" that 3x50 with outside measurements ticked,
to see that the inner measurements of the "hole" is 46.
otherwise some math is required.

What I was asking is just some kind of indicator of what the inner measurements would be, so those numbers could be added to make i.e. a drawer
it could be added alike to the 100mm box reference.

tomse, 2022-03-20:
a better thing would be a tick box "add a drawer" :-P  but for that there would be too many factors. how should the handle look like, does it need an extra frontplate to cover the fingers. should it have separators etc..

I kinda like using your other tools, for that :-)

w1ml33na3rts, 2023-04-05:
I agree about the inner measurements, same for height. I'm trying to design a box for stacks of tiles of 10mm high and 45mm wide and long, but the generator keeps counting the finger joints and thickness of the wood so the inner height is actually around 7mm, same for width and length of the compartments.