florianfesti/boxes #374: DividerTray - last compartment

tomse, 2022-03-20:
Is it possible to have the last angled compartment splitter as close to the back wall as possible.
just like the picture here
https://www.festi.info/boxes.py/DividerTray?language=en

here is the result of my setup, and it's a little difficult to fish up the last item.

[image]

A full width splitter could be useful as well, as you can see in the same pic.
being far from a python dev, I could try and code this splitter, but please bear with me here.

gcollic, 2022-03-21:
Hello :-)

I contributed this box.

> Is it possible to have the last angled compartment splitter as close to the back wall as possible

Yes, it's already possible. It's a section parameter also used in other boxes, there is a link next to it with [more documentation](https://florianfesti.github.io/boxes/html/usermanual.html#section-parameters). Instead of "3*50", you can try "0:3*50:0", and the first and last sections will be of size "0", meaning against the wall.

> A full width splitter could be useful

It's a good idea for some cases! I don't have the time right now to add this option myself, but I could look into it someday if it can wait.

tomse, 2022-03-21:
Nice thank you! :-D 

There is no rush. They are easy to combine by hand

And thank you for contributing this box. It's much appreciated :-D

AllanKustom, 2022-03-21:
How do you get the extra slots in please ? 
Your creation looks great its similar to what I'm trying to make.

florianfesti, 2022-03-21:
Ok, these patches should fix these issues. They are still untested and not on the web instance yet. @tomse can you try them out and report back if everything looks fine, please?

closed 2022-03-21 by florianfesti

tomse, 2022-03-21:
wow.. that was fast. I'll see if I can find time to test it soon (hopefully tonight)

tomse, 2022-03-21:
@florianfesti just tested it.. and result is great. thanks a lot!
I also tested with @gcollic 's suggestion. Which also seems to work wonders. Loving it :-D 

it would be helpful with some help text for gcollic's suggestion,

florianfesti, 2022-03-21:
Actually the "0:" trick is now mentioned in the description text which is displayed below the setting dialog.

tomse, 2022-03-21:
so it is, I missed that :-D  thanks again

commit 3bf4be5e (2022-03-21): DividerTray: Add divider across full width

Resolves: #374