florianfesti/boxes #385: Problem with URL generation

SuksAE, 2022-04-18:
within drawing.py:
self.metadata["url"] returns the correct URL when uisng boxesserver script on local host.
self.metadata["url"] is missing 'boxes.py/' folder when using the server version on https://www.festi.info/boxes.py/

Please investigate and fix.

Thanks
  Klaus

florianfesti, 2022-04-19:
Yeah, looks like this an artifact of the web instance moving into a container. Not quite sure if there is a possible fix or if we need a workaround that patches the missing URL part back in.

SuksAE, 2022-04-21:
Is there some documentation on how to setup such web service? Maybe I can have a look...

florianfesti, 2023-01-11:
Fixed with 0a8dcf0e37880c056d85523cdea97d8c8b166c6e

Not a beautiful solution but should do until I add a config file for the web interface.

closed 2023-01-11 by florianfesti