florianfesti/boxes #391: Gettext failed to extract non-ASCII characters

HaSHsss, 2022-05-13:
In Updating POT files i have issue
`xgettext non-ascii string at boxes/__init__.py:447.`

it is this string `type=ArgparseEdgeType("efFhcESŠikvLt"), choices=list("efFhcESŠikvfLt")`
i think issue in this char "Š" 

Before the latest updates, there were no such problems.

florianfesti, 2022-05-14:
Hmm, works for me. May I ask what OS and Python version you are using?

I thought about adding a utf-8 encoding header to the file but that should not longer be necessary with Python 3 according to the docs.

florianfesti, 2022-05-14:
Or I am missing the point. What exactly are you doing? I tried `python setup.py build` and that worked fine and added the new characters to the .pot file.

HaSHsss, 2022-05-14:
i'm using `python setup.py install --user`
i try `python setup.py build` with no luck. Same issue

My hosting use 
```
Python 3.8.6 (default, Oct 27 2020, 09:13:12)
[GCC 9.3.1 20200408 (Red Hat 9.3.1-2)] on linux
```

florianfesti, 2022-05-14:
OK, looks like xgettext is still using ASCII - at least on RHEL8. Can you try again with 88ea5f8385b824ed8c6fbb458c4e3b053fe8ec0c , please?

HaSHsss, 2022-05-14:
> OK, looks like xgettext is still using ASCII - at least on RHEL8. Can you try again with [88ea5f8](https://github.com/florianfesti/boxes/commit/88ea5f8385b824ed8c6fbb458c4e3b053fe8ec0c) , please?

Yes, it helps. Now everything is fine.

florianfesti, 2022-05-14:
OK, great! Thanks for reporting and testing!

closed 2022-05-14 by florianfesti