florianfesti/boxes #394: Unrecognized arguments in Roudedbox inside Inkscape

mfueto, 2022-05-17:
Everything else works perfect but roundedbox gives this warning.
Inkscape version Inkscape 1.1.1 (3bf5ae0d25, 2021-09-20)
`usage: boxes.exe [-h] [--thickness THICKNESS] [--output OUTPUT]
                 [--format {svg,svg_Ponoko,ps}] [--tabs TABS] [--debug DEBUG]
                 [--reference REFERENCE] [--burn BURN]
                 [--FingerJoint_angle FINGERJOINT_ANGLE]
                 [--FingerJoint_style {rectangular,springs}]
                 [--FingerJoint_surroundingspaces FINGERJOINT_SURROUNDINGSPACES]
                 [--FingerJoint_edge_width FINGERJOINT_EDGE_WIDTH]
                 [--FingerJoint_finger FINGERJOINT_FINGER]
                 [--FingerJoint_play FINGERJOINT_PLAY]
                 [--FingerJoint_space FINGERJOINT_SPACE]
                 [--FingerJoint_width FINGERJOINT_WIDTH] [--x X] [--y Y]
                 [--h H] [--outside OUTSIDE] [--radius RADIUS]
                 [--wallpieces {1,2,3,4}] [--edge_style {f,F,h}]
                 [--top {closed,hole,lid}]
boxes.exe: error: unrecognized arguments: --Flex_stretch=0 --Flex_connection=0 --Flex_distance=0 --Flex_width=0`

florianfesti, 2022-05-22:
This looks like you are mixing newer .inx files with some older version of the boxes.exe binary. May be you have multiple versions installed and Inkscape is using the wrong one. There isn't much that I can do from here. You need to check your installation and update it where necessary.

florianfesti, 2022-06-06:
I don't think there is anything I need to do upstream. Closing to not let this open for ever. Feel free to continue the discussion here anyway or re-open if you think this is an issue with Boxes.py itself.

closed 2022-06-06 by florianfesti