florianfesti/boxes #399: Is it possible to use Boxes.py

c2v4, 2022-06-04:


florianfesti, 2022-06-04:
Boxes.py's kerf correction is done during drawing the parts. As such it can't be allied to existing drawings. There is some code to do kerf correction as a post processing step somewhere in the gear code but it does work on a list of points and not on SVG. So for existing SVG files you may want to use another tool like Inkscape.

closed 2022-06-04 by florianfesti

c2v4, 2022-06-04:
@florianfesti Thanks for so quick response, I could not finish the description so quick ;). As already written, Inkscape is not a solution. I went through the documentation and the second step seemed like something that could be used. Moreover, I saw in issues some regexp manipulation, which is a quite universal way of doing such changes. Anyway, I'll do some manual labor with a file (metal one, not digital ;))

florianfesti, 2022-06-04:
The inset/outset functionality in Inkscape does have some issues. One known workaround is using a line with known thickness, convert the line into an area and then use the outer or inner boarder of that area. But you better ask some Inkscpace folks about that. I did Boxes.py to not use Inkscape...

c2v4, 2022-06-04:
Yeah, I did research on that—in newer Inkscape there is a 'Path Effect' which works great. It offsets both outer edges further out and inner further in. Just plug in half of the laser width (0.08 in my case - kerf 0.16 mm). But that does not account for the rounding near concave corners. That's why I was looking to use Boxes' algorithm—it worked for me flawlessly.