florianfesti/boxes #406: StorageShelf and UniversalBox do not work with Edgetype "t" - Triangle for handle

SuksAE, 2022-06-16:
Tested with default settings, just top_edge parameter set to "t"

TypeTray seems to work fine...

SuksAE, 2022-06-16:
tested on your https://www.festi.info/boxes.py/ server...

HaSHsss, 2022-06-16:
That is known issue
 #305

SuksAE, 2022-06-16:
If in universalbox.py line 78 
`self.h = h = self.adjustSize(h, b, self.top_edge)`
is commented out, it seems to work as it should.
But I admit that I did not test all combinations of top and bottom edge types.

So maybe someone with deeper knowledge about the need of this adjustment should have a look...

HaSHsss, 2022-06-16:
if just comment this line, then the height of the layout is calculated incorrectly

I suggest this option.
```
if self.top_edge == "t":
    self.h = h = self.adjustSize(h, b, e2=False)
else:
    self.h = h = self.adjustSize(h, b, self.top_edge)
```
on line 78

HaSHsss, 2022-06-16:
`RoundedTriangleEdge (t Triangle for handle)` have native default height 150mm
so if set `h = 200mm` it's works fine

I see the easiest way to solve the issue, is just ignore the height of the Triangle handle, but keep considering the height of other `top_edge` views

florianfesti, 2022-06-16:
> I suggest this option.
> 
> ```
> if self.top_edge == "t":
>     self.h = h = self.adjustSize(h, b, e2=False)
> else:
>     self.h = h = self.adjustSize(h, b, self.top_edge)
> ```
> 
> on line 78

this sounds like an easy solution. Guess I have been overthinking this issue.

SuksAE, 2022-06-17:
Guys I must appologise - I was to tired to get this sorted out yesterday, I should have stopped programming earlier.

Now, after a few hours of sleep, I guess I figured it out:

The problem is that the default values of most generators specify an **OUTSIDE** hight of **100mm**. The default heigt of the Triange on the other hand is **150mm**. This can't fit, so the height computes negative and this destorts the output.

I suppose following 2 steps:
1. change the default height value of the triangle (`edges.py line 2359`) to 50mm instead of 150mm to fit into the 100mm default height of most generators
2. include a check at the end of adjustSize() (`__init__.py line 587` ) if the return value is negative and either
- raise an error and stop processing, or
- warn the user about the negative result via text in ANNOTATIONS color within the output file and continue processing with the negative value

HaSHsss, 2022-06-17:
I like this idea, to change default `triangle `height to 50 mm
but it's not enough.

On first pic parameters for
`RoundedTriangleEdge` height =50mm
**Outside** size x/y/h = 100mm
on second same size just for **inside**

So we have 2 completely different boxes turned out,

I think we need to change the `triangle `height to 50 mm
also height calculations for **Outside** 

[image]
[image]

SuksAE, 2022-06-18:
For me, these pictures seems to be OK.

From my understanding, the "**outside**" parameter, as described here: https://florianfesti.github.io/boxes/html/usermanual.html#outside, works as follows:

- when **checked**, the workpiece, including all outside attachment (like handles, hinges, ...), shall fit _within_ a box defined by the given X, Y and H parameters. I use this feature if the workpiece shall fit within a given space like in e.g. a closet or drawer

- when **not checked**, the workpiece shall be able to _contain a payload_ whichs size is defined by X, Y and H. The general outside dimensions of the workpiece (and this includes all external attachments) are not considered in this case. I use this feature if I want the workpiece to store e.g. a tool with a fixed size.

But please correct me if I am wrong...

HaSHsss, 2022-06-19:
> For me, these pictures seems to be OK.
>
> From my understanding, the "**outside**" parameter, as described here: https://florianfesti.github.io/boxes/html/usermanual.html#outside, works as follows:
>
> * [...]

I really don't like such a strong difference in layouts. Anyway, Logic should be standard for all layouts

You're right. 
Probably shouldn't make an exception for 1 layout

closed 2022-06-20 by florianfesti

florianfesti, 2022-06-20:
Yeah, this issue is more complicated than it looks at first glance. If now reduced the default height of the triangle to 50 which makes the issue go away for the default values. But it does not address the underlying issues:

There is no check for values that mess up the generators. This is to some extend by design. Weird values may be useful e.g. for creating a lid from a box with a stackable edge at the bottom and zero height.

But this made into more of a problem by not telling the users the results of the `outside` option. This is an issue in its own though. There is even a ticket for this: #373 

Guess `adjustSize` needs to grow a way to print the new values. Negative values could be colored red to draw attention to situations like this. That way we should be able to get away without checks on the most of the sizes