florianfesti/boxes #409: platonic 20 sides 100mm

YodaMaster18, 2022-06-24:
[image]

Is this working, I have really no idea :-)

florianfesti, 2022-06-24:
This is working but doesn't show the whole piece. Not sure if this is an issue with posting or the picture was like that from the beginning. May be try a picture from further away. I can cut it to size later.

YodaMaster18, 2022-06-26:
[image]

I think this is better, thnx

florianfesti, 2022-06-26:
Thanks for the extra effort. I will not update the web instance right away but the image will show up as soon as I will.

closed 2022-06-26 by florianfesti

YodaMaster18, 2022-06-26:
Thank you!Vriendelijke groeten,Jochem Douwes Van Blommesteinstraat 22614 ER Delft 06-28411405

commit 73a46460 (2022-06-26): Platonic: Add image of Icosahedron

Thanks to https://github.com/YodaMaster18 for the image!

Resolves: #409