florianfesti/boxes #414: Request - bird house

KittyDreadful, 2022-07-18:
[image]

I'd like to suggestion a roof design option - please see the attached picture.  

Thanks!

florianfesti, 2022-07-18:
You can use a normal FingerJoint edge (f) for this. Just choose a large enough values for the `finger` and `space` settings.
For the roof use a FingerHole edge (h).

See [AllEdges](https://festi.info/boxes.py/AllEdges?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=10&FingerJoint_play=0.0&FingerJoint_space=10&FingerJoint_width=1.0&Stackable_angle=60&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&Hinge_grip_percentage=0&Hinge_outset=0&Hinge_pinwidth=0.5&Hinge_style=outset&Hinge_axle=2.0&Hinge_grip_length=0&Hinge_hingestrength=1&Lid_angle=90.0&Lid_hole_width=0&Lid_second_pin=0&Lid_second_pin=1&Lid_spring=both&Lid_style=rectangular&Lid_surroundingspaces=2.0&Lid_edge_width=1.0&Lid_extra_length=0.0&Lid_finger=3.0&Lid_play=0.05&Lid_space=2.0&Lid_width=1.0&Click_angle=5.0&Click_bottom_radius=0.1&Click_depth=3.0&Flex_stretch=1.05&Flex_connection=1.0&Flex_distance=0.5&Flex_width=5.0&x=100&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&inner_corners=loop&burn=0.1&render=1) for examples with `finger=10` and `space=10`.

florianfesti, 2022-07-18:
Ah, f***k it.... Here you go: 
https://festi.info/boxes.py/BirdHouse?language=en

Be aware that you are liable to test cut the bird house and provide an image.

florianfesti, 2022-07-18:
OK, few more fixes in 0938d12aa0ca76453e05ac4e9ee109f211971c34

The full code is at https://github.com/florianfesti/boxes/blob/master/boxes/generators/birdhouse.py

KittyDreadful, 2022-07-18:
THANK YOU SO MUCH!

KittyDreadful, 2022-07-19:
There's something off...

1. It looks like the line on the bottom of the front and back aren't connecting correctly.   Can you see where the line extends to the left a little?
[image]

2.  I think that this is causing the tabs on the base to be off center.  There's a gap.
[image]

florianfesti, 2022-08-12:
Did that fix the issue?

commit 122d3659 (2023-02-26): BirdHouse: Add overhang parameter

to allow making the roof bigger and better looking.
Declare stable by moving to "Misc"

Resolves: #414
closed 2023-02-26 by florianfesti