florianfesti/boxes #416: Drillbox Generator improvement

lost-hope, 2022-07-27:
Hello, 
first of all great project overall. used it on my private laser cutter and when we got one in the company i pulled it up in front of the sales guy, and he was in awe. 

I wanted to make a drillbox for some drills (obviously)  and saw that you can enable the stagable edge for the bottom.

is it possible to have that option for the top edge as well? that way i could have multiple boxes of the same size and stack them to safe some space when not needed
Generator URL:
https://festi.info/boxes.py/DrillBox?language=en

closed 2022-07-28 by florianfesti

florianfesti, 2022-07-28:
Yes, this is one of the older box generators that do not yet have some of the improvements of some of the more modern ones.

lost-hope, 2022-07-28:
Thanks.
I will upload a photo once i have cut and assembled it

commit 80765666 (2022-07-28): DrillBox: Add support for top_edge parameter

with limited selection of edge types (compared to UniversalBox)

Resolves: #416