florianfesti/boxes #417: Planetary2 Issue

lost-hope, 2022-07-27:
in the Generator for the Balanced force Difference Planetary Gear (https://festi.info/boxes.py/Planetary2?language=en) seems to be an issue where the holes seem misplaced with the default settings: 
[image]
[image]
[image]
[image]

florianfesti, 2022-07-28:
Well, this thing should probably be in the Unstable group. I doubt this is the only issue...

florianfesti, 2022-07-31:
Oh, I fixed the modus handling at some point. So the default is a bit large now. If you lower the modulus to 1or 1.5 the spokes are gone.

florianfesti, 2022-09-17:
OK, I moved the generator to "Unstable". This needs more work to get really working. And I currently don't have the time for that. I may get back to it later or some one else can pick this up.

commit 93a4fad7 (2022-09-17): Planetary2: Move to Unstable as it is not yet working properly

The middle gear set can't fulfill the planetary gear set condition
(ring = sun + 2 * planets)
May have more issues.

Resolves: #417
closed 2022-09-17 by florianfesti