florianfesti/boxes #422: TypeTray - inconsistent usage of lids.py/topEdges return values

SuksAE, 2022-08-20:
There is an inconsistency in the usage of topEdges function within TypeTray.
When reverse engineering the source code of topEdges function the return values are in the order left/back/right/front edge.

a. this should be documented (at least in soucecode) so that no reverse engineering is needed
b. TypeTray.py should be fixed accordingly
     e.g change line 103 to `t2, t1, t4, t3 = self.topEdges(self.top_edge)`

florianfesti, 2022-10-01:
Guess is this fixed with 93a4cad4f49f3d7043bd288e08b216da64dfa87a

I added a7fb56230698d0e99867bbfffb8d4b204e80cecf , 3a06aa1d62b5c0fa8b809b9dc4f275c0408930f7 and c4b0971d039e718a970037d45372bf82b48e538a to clean that up even a bit more.

In theory we should also rename the top edge variables in

- https://github.com/florianfesti/boxes/blob/master/boxes/generators/drillbox.py
- https://github.com/florianfesti/boxes/blob/master/boxes/generators/magazinefile.py
- https://github.com/florianfesti/boxes/blob/master/boxes/generators/storageshelf.py
- https://github.com/florianfesti/boxes/blob/master/boxes/generators/ubox.py
- https://github.com/florianfesti/boxes/blob/master/boxes/generators/universalbox.py

but I just can't be bothered right now. Closing as fixed!
Thanks for the patch!

florianfesti, 2022-10-01:
Also the order of the edges is a bit odd. front, right, back, left would be a lot more in line with all the other `edges` parameters for walls. But that is only really easy to change after the generators above have their variables renamed.

closed 2022-10-01 by florianfesti