florianfesti/boxes #423: Adding label results in assertion

SuksAE, 2022-08-20:
When adding a label to line 163 and/or 166 of TypeTray.py

```
            self.trapezoidSideWall(
                y, h, h+bh, [b, "h", "e", "h"],
                radius=self.radius, callback=[self.yHoles, ], move="up", label="bad")
```

this results in an assertion:

> Exception during rendering:
> Traceback (most recent call last):
>   File "C:\Daten\Projekte\Python\boxes_SuksAE\scripts\boxesserver", line 517, in serve
>     box.close()
>   File "C:\Daten\Projekte\Python\boxes_SuksAE\scripts\..\boxes\__init__.py", line 715, in close
>     self.surface.finish(self.inner_corners)
>   File "C:\Daten\Projekte\Python\boxes_SuksAE\scripts\..\boxes\drawing.py", line 474, in finish
>     extents = self._adjust_coordinates()
>   File "C:\Daten\Projekte\Python\boxes_SuksAE\scripts\..\boxes\drawing.py", line 61, in _adjust_coordinates
>     self.transform(self.scale, m, self.invert_y)
>   File "C:\Daten\Projekte\Python\boxes_SuksAE\scripts\..\boxes\drawing.py", line 73, in transform
>     p.transform(f, m, invert_y)
>   File "C:\Daten\Projekte\Python\boxes_SuksAE\scripts\..\boxes\drawing.py", line 109, in transform
>     assert(not self.path)
> AssertionError

florianfesti, 2022-08-20:
Yep, easily reproduced. There is something very wrong in the backend. May take a bit to figure out.

Generally the assert is for parts that have been drawn but have not been finalized by `ctx.stroke`. The label messing zthis up make the error even more mysterious.

SuksAE, 2022-08-22:
strange behavior in __init__.py:

**1st try** - function trapezoidSideWall():
changing line 2621 from
`        self.move(overallwidth, overallheight, move, label)`
to
`        self.move(overallwidth, overallheight, move)`
--> no assertion, everything works as expected

BUT:
**2nd try** - function move()
changing lines 1184 and 1185 from
```
            if self.labels and label:
                self.text(label, x/2, y/2, align="middle center", color=Color.ANNOTATIONS, fontsize=4)
```
to
```
#            if self.labels and label:
#                self.text(label, x/2, y/2, align="middle center", color=Color.ANNOTATIONS, fontsize=4)
```
the assertion persists although no label is generated...

Do you have any ideas?

florianfesti, 2023-01-16:
OK, this took a bit longer to look into from the right angle. There's no great mystery here only label not actually landing in the label parameter of the move method... *doh*

commit 971c20be (2023-01-16): trapezoidSideWall: Fix call to move

resulting in an assert error as the label was used as "before" parameter
which messes up the whole magic inside .move(). As a result the .stroke() method was not called - leading to unprocesses paths - triggering the assert.

Resolves: #423
closed 2023-01-16 by florianfesti