florianfesti/boxes #427: Accessibility navigation issues for screen reader users

funkonaut, 2022-08-31:
There are some issues in the web interface that affect non-visual access to the web page. My pull request attempts to solve them by labeling form fields for screen readers to announce the input elements and making clickable headings keyboard navigable. There is some redundancy with aria-labels as well to improve access by announcing the help text as well as the name when form fields are focused.

Using [this accessibility checker.](https://wave.webaim.org/)

funkonaut, 2022-09-04:
Issues:
-Table elements that wrap the input forms for each box generator announces extra information such as column and row name according to [W3 WAI reference ](https://www.w3.org/WAI/tutorials/tables/) tables should not be used for layout purposes but fixing this would require a large rewrite of the python/html adding role="presentation" tag provides a work around solution 
-Form fields are not labeled and thusly are not read by screenreaders.
-Clickable headings have device dependent event handlers and no tab index, as they are clickable they should be tab navigable and event handlers should be device independent, so functionality is accessible to both mouse and keyboard users

@HaSHsss do you have any additional issues you would like to add?

florianfesti, 2022-09-10:
#480 should have solved all of the above. Please re-open or file an new issue if there is anything else we can do to improve accessibility.

Closing.

closed 2022-09-10 by florianfesti