florianfesti/boxes #429: HolePattern creates holes other than specified

Bambusbjoern, 2022-09-02:
Seemingly depending on fill_pattern, space_between_holes, space_to_border the size of holes created by HolePattern doesn't match the size set with hole_min_radius and hole_max_radius.

E.g.:
https://www.festi.info/boxes.py/HolePattern?fillHoles_bar_length=50&fillHoles_fill_pattern=square&fillHoles_hole_max_radius=9.25&fillHoles_hole_min_radius=9.25&fillHoles_hole_style=round&fillHoles_max_random=1000&fillHoles_space_between_holes=2.8&fillHoles_space_to_border=2.8&x=184&y=119&shape=rectangle&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&inner_corners=loop&burn=0.1&render=1

https://www.festi.info/boxes.py/HolePattern?fillHoles_bar_length=50&fillHoles_fill_pattern=hex&fillHoles_hole_max_radius=9.25&fillHoles_hole_min_radius=9.25&fillHoles_hole_style=round&fillHoles_max_random=1000&fillHoles_space_between_holes=2.8&fillHoles_space_to_border=2.8&x=184&y=119&shape=rectangle&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&inner_corners=loop&burn=0.1&render=1

florianfesti, 2022-09-10:
@SuksAE Do you want to take look at this?

SuksAE, 2022-09-10:
My first thought is that maybe this can be reduced to a "lack of documentation" issue.

Maximum hole size is used unfiltered only in "random" mode. In "square" or "hex" mode the default hole size is clalculated to choose an "optimum" size so that an integer number of shapes fit in a row. 
The size of the drawn holes is always between the maximum hole size and minimum hole size given in the settings.

see init.py line 1689:
```
        elif pattern in ("square", "hex"):
            # use 'optimum' hole size
            max_radius = min(max_radius_x, max_radius_y)
```

I hope this solves the mystery...

Bambusbjoern, 2022-09-11:
In the linked examples both minimum and maximum hole size is 9.25 though.

SuksAE, 2022-09-11:
You are right.

Minimum hole size is actually only checked if there is an "abnormality" in a line of shapes. In this case the algorythm tries to "squeeze" in smaller holes. But as there are no abnormalities in the given example (shape is rectangle --> all lines of equal length, always the same integer number of holes), minimum hole size is irrelevant here...

For visualization of minimum size usage try:
[min radius = max radius](https://www.festi.info/boxes.py/HolePattern?fillHoles_bar_length=50&fillHoles_fill_pattern=square&fillHoles_hole_max_radius=9.25&fillHoles_hole_min_radius=9.25&fillHoles_hole_style=round&fillHoles_max_random=1000&fillHoles_space_between_holes=2.8&fillHoles_space_to_border=2.8&x=184&y=119&shape=ellipse&thickness=3.0&format=svg&tabs=0.0&debug=0&debug=1&labels=0&labels=1&reference=0&inner_corners=loop&burn=0.1&render=1)
in comparison with
[min radius << max radius](https://www.festi.info/boxes.py/HolePattern?fillHoles_bar_length=50&fillHoles_fill_pattern=square&fillHoles_hole_max_radius=9.25&fillHoles_hole_min_radius=1&fillHoles_hole_style=round&fillHoles_max_random=1000&fillHoles_space_between_holes=2.8&fillHoles_space_to_border=2.8&x=184&y=119&shape=ellipse&thickness=3.0&format=svg&tabs=0.0&debug=0&debug=1&labels=0&labels=1&reference=0&inner_corners=loop&burn=0.1&render=1)

Please note that:
- Red lines in the exampes is debugging information to visualize the shape to be filled ("ellipse" in these cases)
- If you want try your own parameter set, you can get back to the configuration page if you replace "render=1" with "render=0" in the URL.

florianfesti, 2022-09-27:
When we take back a step the issue here is that the new hole pattern code is not yet general enough for all use cases. In the medium term it should replace the legacy `hexHoles` code already in the code base. For that we need a few more knobs. Adjusting the hole size is a feature  that is actually even on the todo list of the hexHoles. But many use cases require fixed hole sizes as the holes have other purposes than being decorative or letting air through.

That being said removing the hexholes generator may have been a bit premature. Otoh this highlights nicely that there is still work to be done.