florianfesti/boxes #430: Live 2D preview in web interface

pvh, 2022-09-03:
Instead of toggling back and forth between a preview and the form, it would be nice to have an inline view of the resulting document. 

If you agree this is a useful feature I could take a shot at implementing it.

florianfesti, 2022-09-10:
Yes, a live preview would be really nice. Screen space is pretty limited though. But I am really not a web/UI person and someone more talented than me might be able to make this work.

The live preview should probably be closed and only be opened on click/command. Not sure how my little server will react if requests go up by a factor of 20. But that's something we can see as soon as we get there.

florianfesti, 2023-02-07:
With the performance issues we had on the server lately I don't think we can afford a live preview. Inkscape can do something like that for the plugin. Closing.

closed 2023-02-07 by florianfesti