florianfesti/boxes #431: Request - Bread box

mauroinagoki, 2022-09-10:
hi,

I would like to have a breadbox generator that is similar to your shutterbox

[image]

Source: https://vectezia.com/laser-cut-living-hinge-bread-box-with-sliding-lid-vector-cdr-file/

florianfesti, 2022-09-23:
The first version is available in the Unstable category.

By requesting this new generator you agreed to test cut it until it works properly (which is probably won't right away) and to provide a nice sample photo once it does.

https://festi.info/boxes.py/BreadBox?language=en

florianfesti, 2022-09-23:
Please only cut this very small until it is verified that everything fits together.

florianfesti, 2022-12-06:
Did everything fit? Do you have a picture?

commit d3df27b2 (2023-01-13): BreadBox: Declare stable and add sample image

Thanks to Ed Nisley <https://github.com/ednisley> for the picture!

Resolves: #431
closed 2023-01-13 by florianfesti