florianfesti/boxes #434: Request: SlatWallEdges with more configurability

jan-eckhoff, 2022-09-17:
I have a board game table with a rail system that is functionally working like a slatwall, but has other measurements.

Specifically, I need to adjust the vertical height of the hook (right now, it just slides out because it's too small), and although it seems to be fine with my table, it would be nice to set the horizontal distance between the inner edge of the hook and the back of the box.

florianfesti, 2022-09-17:
Any suggestion for the parameter name? Some idiot has used `hook_depth` for something else already...

jan-eckhoff, 2022-09-17:
Hi & wow that was fast :-D
I would have looked for something like hook_height for the first parameter, and maybe hook_distance or something like that for the second one.

closed 2022-09-17 by florianfesti

florianfesti, 2022-09-17:
You are lucky. I really don't want to start a 3rd page of open issues.

jan-eckhoff, 2022-09-18:
Thanks! I‘m quite new to GitLab, so maybe I missed it, but did you also implement the hook_height parameter?
Also, will this deploy to the web interface at some point, or would I have to use the script directly to use those parameters?

florianfesti, 2022-09-22:
Yeah, I closed this a bit too early. Kids, don't drink and push...

Anyway, added more parameter and updated the web instance.

commit 15933311 (2022-09-17): SlatWallSettings: Add hook_distance parameter

Resolves: #434