florianfesti/boxes #436: Request - Card Box adjustable dividers

agreen, 2022-09-23:
I would like to see some card box for each compartment. I envision this as removing the swoop the compartment dividers have now, and creating evenly spaced finger joints along the top, that allow for a removable, adjustable divider to be slid in.  The adjustable divider would be the width of the compartment, with one finger joint on each side, only on the top.  There could or could not be a cutout on the top of each adjustable divider to allow room for your fingers to get to the cards.

Some context; My son collects pokemon cards, and sections them off based on attributes.  subdivisons as I describe would allow him to do that better, and just placing a divider "card" within the current box design, the card can slide down, along the with the entire row of cards.

agreen, 2022-09-23:
For reference, I envision something like this - https://www.instructables.com/How-to-Design-and-Laser-Cut-a-Game-Card-Storage-Bo/

but it would be helpful if the dividers could be placed parallel, instead of needing to stagger them. I think that can be accomplished just by cutting the finger on the adjustable divider to half the thickness of the material

gcollic, 2022-09-23:
Hello!

I'm not the author of this project, but I contributed a box just like that : https://www.festi.info/boxes.py/DividerTray?language=en 🎉 

It's highly configurable. You can have a bottom or not, etc.

It lacks a lid, but there is plenty of options here to generate one.

> it would be helpful if the dividers could be placed parallel, instead of needing to stagger them. I think that can be accomplished just by cutting the finger on the adjustable divider to half the thickness of the material

I did experiments with that (I just saw that the picture I contributed even shows that, even when the box doesn't support it in the parameters 😞 ). But the results were not great when the cards are heavy. Since then I tried a double width wall (done with a row of 0 width) and I found it much better when you absolutely want modular dividers which can be parallel.

We also now have a divider generated which spans across all the rows. It's by far the best option for parallel dividers, if afterwards you don't need to be able to remove the dividers individually row by row ;-)

Let us know what you think of it, and if it fits your needs!

florianfesti, 2022-09-23:
What about just adding another row of dividers with the side tabs only being half a thickness as suggested above? That's easy to do and then everyone can pick and choose. That's probably even better than adding another parameter.

florianfesti, 2022-09-23:
Well, just adding them is probably quicker than having a lengthy discussion. I am still interested if this works out for what you want to do.

closed 2022-09-23 by florianfesti

agreen, 2022-09-23:
I think this divider would work for what I want, however I was looking for something with a lid as well. I think I could cut this without the bottom, and glue it into one of the other box generators, or generate a lid for it using a shallow box as a lid.  A few different options there.

commit 41ecd619 (2022-09-23): DividerTray: Add dividers with half width tabs

This allows having dividers side by side.

Resolves: #436